sinara-hw/AUX_PSU #1: [RFC] Aux Power Supply

dtcallcock, 2020-06-25:
In my experience, lab setups rapidly acquire an armada of bench PSUs supplying the same old low voltage DC rails for misc homebrew electronics boxes.  Generally these PSUs are bulky, expensive, have much higher current spec than required, and sometimes have knobs on the front that get accidentally knocked with bad consequences.

It would be nice if there was a simple 3U, 4HP card capable of producing common low-current DC voltage rail combinations.  This board is fairly simple so the main question is probably whether enough people would find it useful and purchase it to make it worth while.

### Inputs
12V - via front panel barrel, rear Molex, and EEM

### Outputs
Voltage rail 1 - selectable between 3.3, 4, 5 and 6V - 1A
Voltage rail 2 - selectable between 5, 9, 12, 15, 18 V - 500mA
Voltage rail 3 - selectable between  -5, -9, -12, -15, -18 V - 500mA

Rotary switches to select voltage on each rail independently.  With a hex switch we can have 16 voltages per rail but a few obvious ones are listed above. 

Rails generated using SMPS followed by LDO.

Specs in line with typical decent bench PSU.  Example spec taken from Keysight E3630A as a guide: 

Load and line regulation: < 0.01% + 2 mV
Ripple and Noise (20 Hz to 20 MHz): < 350 μVrms, < 1.5 mVpp
Temperature coefficient per C: < 0.02% + 1 mV
Output Drift: Less than 0.1% + 5 mV
Transient response time: < 50 μsec following a change in output current from full load to half load for output to recover within: 15 mV

Overcurrent indicated by LED.  Rail would continue to operate (with no promises on spec) until thermal limit exceeded.  At which point thermal protection LED would come on and board would shut down.  

### Connectors
M8-4 - A compact 4-pin connector - could fit 4/5 on panel.  

Convenient pigtails [are available](https://www.digikey.com/product-detail/en/phoenix-contact/1406203/277-12862-ND/4291415) for ~$10.  Shielded versions also available for $20.

### Indicators
Overcurrent LED for each rail
Thermal protection LED

Edit: updated this top post to reflect discussion below

michalgaska, 2020-07-14:
I found other LDOs that are more suitable for our project. LT3089 and LT3090. 
Both have a programmable current limit and output current monitor. LT3089 has also a junction temperature monitor.
I made output noise simulation for both devices:
LT3089:
There is no spice model for LT3089. Instead of a model for LT3089, I used the LT3081 model which is very similar. 
[image]

LT3090:
[image]

gkasprow, 2020-07-16:
looks good! 
@michalgaska What do you think by "more suitable" ?

closed 2021-05-22 by gkasprow

gkasprow, 2021-05-22:
finished board
[image]

jordens, 2021-08-02:
@gkasprow Is this ready for production or has there been a run already?

gkasprow, 2021-08-02:
TS is currently producing it AFAIK