sinara-hw/AUX_PSU #2: Powering options/EEM

dtcallcock, 2020-06-28:
Options for getting 12V to power the board:

- Barrel connector on the front panel
- Molex connector on the back of the board
- EEM
- PoE

The first two seem like they definitely should.  I think the EEM is worth doing (like on clocker).  I don't think the PoE is, but mention it as it's become pretty standard in Sinara.

If EEM is implemented do we want the following features:
- A diode (ideal diode?) on the 12V line like Clocker to prevent [these kind of issues](https://github.com/sinara-hw/Stabilizer/issues/72).
- The usual EEPROM for identification via I2C (this wasn't done on Clocker).
- An I2C IO extender for monitoring overtemp/overcurrent/pgood etc.
- EEM to be located correctly for the CPCIS adaptor.

gkasprow, 2021-05-22:
I added the ideal diode protection

gkasprow, 2021-05-22:
we also have I2C ADC that monitors temperatures, currents, and voltages.
we don't have LDO PG to monitor.

closed 2021-05-22 by gkasprow