sinara-hw/AUX_PSU #3: LDOs

dtcallcock, 2020-07-14:
I found other LDOs that are more suitable for our project. LT3089 and LT3090. 
Both have a programmable current limit and output current monitor. LT3089 has also a junction temperature monitor.
I made output noise simulation for both devices:
LT3089:
There is no spice model for LT3089. Instead of a model for LT3089, I used the LT3081 model which is very similar. 
[image]

LT3090:
[image]

_Originally posted by @michalgaska in https://github.com/sinara-hw/AUX_PSU/issues/1#issuecomment-658224245_

dtcallcock, 2020-07-14:
I made this its own issue to start breaking things out of the original RFC.

Compared to the TPS7A47-Q1, the LT3089 has somewhat worse specs:
- Output voltage noise: 27uVrms vs 4.2uVrms
- PSRR @ switcher freq: 40dB vs 60dB
- Dropout voltage: 1.5V vs 500mV

The latter would increase the power dissipated on this card by a couple of Watts at full draw.

However these aren't show stoppers (noise would still be better than typical bench PSU)  and might be a price worth paying for the convenient current monitor, temp monitor, and  programmable current limit.

gkasprow, 2020-07-16:
How precise is this current monitor? Of course it is simpler than just using the external sense resistor + opamps.

closed 2021-05-22 by gkasprow