sinara-hw/AUX_PSU #4: schematic review

gkasprow, 2020-09-26:
- [ ] NMOS won't work for a negative rail. Use PMOS + level shifters
- [ ] add LC filter between SMPS and LDO, they have not sufficient PSRR for switching frequency
- [ ] resistors at the decoder outputs are not needed
- [ ] now the inactive switch positions result with max output voltages
- [ ] use 8-position switches like[ this one](https://www.digikey.com/product-detail/en/S-8011/563-1054-ND/948365?utm_campaign=buynow&utm_medium=aggregator&curr=usd&utm_source=octopart). There is 3D model on Digikey.
- [ ] assign all 8 positions, 3V and 6V is often needed. The rest of them must give known value.
- [ ] alternatively, one can use [standard](https://pl.mouser.com/ProductDetail/NKK-Switches/NR01105ANG13-2A?qs=EU6FO9ffTwcM3pHNV2SdFA%3D%3D) 1:6 switch, but it would need a tiny RA adapter - a piece of PCB and RA 2.54 connector. But they are nearly 10$ per piece. However, they would last longer.

gkasprow, 2020-09-26:
@dtcallcock @jbqubit @hartytp can you have a look at the schematics?
@michalgaska please place a PDF version to make review easier.

dtcallcock, 2020-10-21:
Any joy on the pdf?

jbqubit, 2021-02-26:
What's the status of this board? Searching for a schematic PDF still brings more tears than joy.

gkasprow, 2021-02-26:
@jbqubit here is the schematic. The output connector signal assignment will change.
[Aux_PSU.PDF](https://github.com/sinara-hw/AUX_PSU/files/6052347/Aux_PSU.PDF)

gkasprow, 2021-02-26:
It goes to production together with SHuttler boards

michalgaska, 2021-02-27:
I committed the current design files. 
Schematic: 
[AuxPowerSupply.pdf](https://github.com/sinara-hw/AUX_PSU/files/6053862/AuxPowerSupply.pdf)

dtcallcock, 2021-03-05:
- It's a bit odd one of the OC LEDs is green, not red.  I guess it would add a BOM line though.

[image]

- I think it'd be much more useful to have all 3 voltages plus ground on each output connector instead of one voltage on each connector.

- The rear power connector is barrel right now.  I thought the Molex type was the preferred population option for internal power connectors though.

gkasprow, 2021-03-05:
> I think it'd be much more useful to have all 3 voltages plus ground on each output connector instead of one voltage on each connector. 
>The rear power connector is barrel right now. I thought the Molex type was the preferred population option for internal power connectors though.
>It's a bit odd one of the OC LEDs is green, not red. I guess it would add a BOM line though.

are you looking at the right schematic? This was fixed some time ago...

dtcallcock, 2021-03-05:
My bad - clicked wrong link above.

- My comment about the Molex still stands though - shouldn't it have a dual footprint like on Stabilizer:
[image]

- Can we get an annotation about the OC setpoints and programmed regulator current limits?  Right now it looks like the positive rails are set to 600mA limit with 500mA OC threshold but the negative rail is 500mA/500mA.

- As the second positive rail is typically used for V_DD, would 1.8 and 2.5 options be more useful than 15 and 18?  It might be nice to trade the lower voltages for more current in this way too.  Could the limit be raised on this rail if the higher voltages aren't an option?  It looks like going above 800mA would involve paralleling a second LDO so that's perhaps not worth it.

- Why does the temperature sensor only shut off the negative rail?

- Can a 0R be added between the M8-4 shield and ground?  That would allow one to depopulate this if they wanted the shield connected at the load only.  Without being able to see the M8-4 part number, I cannot tell if the shield is likely to be isolated from the front panel.  If it isn't that probably rules this use case out.  However in that case, I think you still want the 0R as you need it to be able to isolate the front panel.

gkasprow, 2021-05-22:
I implemented all suggestions

closed 2021-05-22 by gkasprow

dtcallcock, 2021-05-24:
Can you make a release with pdf if you have a min?

gkasprow, 2021-05-25:
done