sinara-hw/AUX_PSU #10: Stock problems

pkozakiewicz, 2021-08-31:
I would like to address some availability problems:

LT3090HMSE#PBF - can we use different temperature range version? (EMSE, IMSE),
LT8362IDD#TRPBF - same as above (EDD),
LM75AIMM/NOPB - is there any 1:1 replacement?

gkasprow, 2021-08-31:
Yes, you can use another temp range.
LM75BIM-5/NOPB would work

pkozakiewicz, 2021-09-23:
These are another two shortages:

1SMB5916BT3G
3059Y-1-254LF

martafafara, 2024-10-09:
Most of above parts had only temporary stock problems, except LM75AIMM/NOPB which is obsolete. TI recommended replacement: LM75AIMMX/NOPB.
Detailed analysis of BOM revealed additional necessary changes:
CD4028BM-> CD4028BM96
GRM219R61E106KA12D ->TMK212BBJ106KGHT

closed 2024-10-16 by martafafara