sinara-hw/AUX_PSU #14: LT3089IFE - wrong values of voltage setting resistors

mkiepiela, 2021-12-07:
All the voltage setting resistors' values at both positive LDOs are wrong (two times to low).

gkasprow, 2023-01-07:
It's not only resistor values that are wrong. The MOSFET Iq causes slight shift and additonal temperature dependence.

gkasprow, 2023-01-07:
[image]

gkasprow, 2023-01-07:
we can use CSD15380F3

gkasprow, 2023-01-07:
For negative channel we use BSS184 which has followin Idss
[image]

What's interesting, when I heat the transistor area, the output voltage gets higher by a few mV

gkasprow, 2023-01-07:
The effect I observed was caused by the resistors tempco.

martafafara, 2024-10-15:
Resistors values corected. 
Transistors for positive rails are BSS138LT1G. According to datasheet max. leakage current do not exceed 20nA, what gives additional current at the level of 140nA for all transistors on rail. The error level due to Idss is then up to 0,28% what is negligible.
[image]

closed 2024-10-15 by martafafara