sinara-hw/AUX_PSU #17: production status

jordens, 2022-03-30:
According to @pkozakiewicz the design is pretty much non-functional as it is currently. I think the issues discovered have been filed.
@dtcallcock you were originally pursuing this project. Did you have a look at the board?
@gkasprow is there a way to revise/fix/redesign it?

michalgaska, 2022-03-30:
I am going to fix all the issues with the design.

jordens, 2022-03-30:
That would be great!
Did you manage to rework a v1.0 board to make parts of the design work?

pkozakiewicz, 2022-03-30:
@gkasprow has one board from me and is still working on the fix.

jordens, 2022-05-03:
@michalgaska how is the progress?

jordens, 2022-05-25:
@michalgaska ping

pkozakiewicz, 2022-05-26:
@michalgaska In addition to releasing new revision, please also prepare a patch for existing v1.0 boards.

michalgaska, 2022-05-30:
@jordens I am working on the existing board to fix all the issues. Once I finish that I will make changes to the design.

jordens, 2022-08-23:
@michalgaska How is the work going?

michalgaska, 2022-08-25:
@jordens I fixed all the issues and the board works.  The only thing I need to do is tune the resistor value to have a better over-current indication and check over-temperature protection. Vacation is ending so it is a good time to go back to this project and finish it.

jordens, 2022-09-12:
@michalgaska Thanks for the update! Did you have a chance to look at the issues you described?

jordens, 2022-12-07:
@michalgaska @gkasprow How can we proceed here?

pkozakiewicz, 2022-12-08:
I still have not received a clear instruction what ought to be done with these boards. I count on @michalgaska to send me a detailed description.

jordens, 2022-12-08:
@gkasprow Thanks. Having had a look at the issues for the v1.0 and hearing what @pkozakiewicz said, I'd be very much interested in v1.1 prototypes.

michalgaska, 2022-12-11:
@pkozakiewicz I will prepare it on Monday.

jordens, 2022-12-20:
@michalgaska ping

pkozakiewicz, 2022-12-27:
@gkasprow @michalgaska Any progress here?

gkasprow, 2023-01-07:
I made 2 pieces running.
I tested:
- output voltages switching. I used 250ppm 1% resistors which are probably not sufficient for this purpose
- current limit LEDs
- current limiting
- overtemp protection
- LDO dropout control. It changes +/-20mV  at max load
- ADC input values. I didn't check the I2C readout
- ripples below 10mV, no traces of DC/DC switching signal when no load. Ripples come mostly from ground loop. When full load is applied the ripples are ~10mV and they come from DC/DC converter modulation
[image]

gkasprow, 2023-01-07:
I added 68nF between C and E of T17 and 1nF in parallel to C43 but the difference is very subtle
[image]

gkasprow, 2023-01-07:
The output voltage was set to 12V and I used 22Ohm load

gkasprow, 2023-01-07:
Another positive channel has the same issue with DC/DC. Vout=12V, Rl=22Ohm
[image]

gkasprow, 2023-01-07:
negative LDO does not have such issues
[image]

gkasprow, 2023-01-07:
It looks like we need an LC filter between DC/DC and LDO. The FB is not suited for this purpose.

gkasprow, 2023-01-07:
Idle power consumption is 60mA

gkasprow, 2023-01-07:
Anyway, the DC/DC converter should not produce such 700mV ripples.

gkasprow, 2023-01-07:
Actually, now the FB of the DC/DC converter is quite complex because it involves LDO transimpedance and also depends on LDO load.

gkasprow, 2023-01-07:
I finished the debugging session for today. The following things to check:
- the negative rail has twice the higher DC/DC converter output capacitance (20uF vs 10uF)
- replace L11 with 2.2uH and add 10uF tantalum parallel to C66 . That would make 23kHz LPF. The FB needs to be there to filter HF components of the DC/DC coverter

jordens, 2023-01-07:
Oh I hadn't noticed that the LDO is inside the SEPIC feedback loop and feedback is for constant dropout. Certainly innovative.

gkasprow, 2023-01-07:
Actually, it's a known technique from AFAIR ADI app note

pkozakiewicz, 2023-01-24:
@gkasprow Any progress here? When should we expect these boards back?

gkasprow, 2023-02-07:
> 
>     * the negative rail has twice the higher DC/DC converter output capacitance (20uF vs 10uF)

adding 10uF helped. 47uF did a job.

> 
>     * replace L11 with 2.2uH and add 10uF tantalum parallel to C66 . That would make 23kHz LPF. The FB needs to be there to filter HF components of the DC/DC converter

I used 10uH inductor in series with FB. That significantly reduced the ripples.

> 
>     * the negative channel 200k FB resistor was not changed

The value is critical because after adding 120k in parallel to 200k the ripples increased significantly. 

120k seems to be fine resulting in 2V dropout