sinara-hw/AUX_PSU #25: iso-SPI

gkasprow, 2024-01-15:
We must build several custom front-ends for photonic sensors, like superconducting nanowire detectors for space-to-ground QKD.
The idea is to reuse this board and add 3 ISO-SPI transceivers and DIOT support.
In this way, with a single AUX_PSU module and primitive AFE boards (just SPI ADC/DAC), we can build a complex system with minimal HW RnD.
So, the idea is to add a mezzanine with [LTC6820](https://www.analog.com/en/products/ltc6820.html), LVDS transceivers, isolation transformers and [SPE connectors](https://www.mouser.pl/ProductDetail/Phoenix-Contact/1163797?qs=DPoM0jnrROVUdkI1ZYBH3w%3D%3D&mgh=1&vip=1&gad_source=1&gclid=CjwKCAiAzJOtBhALEiwAtwj8tmuNis1pGjHkZxSd9ASxfZz7sCrSd19tx1krNMhOuOITj4Iq5-q8tBoC0S0QAvD_BwE). 
Is integrating the power supply and isolated digital signals in one M10 connector a good idea?
Alternatively, we could replace M10 with RJ45 connectors and use standard Ethernet cables for power supply and communication.
RJ50(10P10C) could also be a good option to avoid plugging Ethernet or LVDS cable into a 15V supply.

kaolpr, 2024-10-15:
Does this idea still have a use case?

gkasprow, 2025-05-16:
If it's not a problem, we could add a connector and mounting holes for ISO SPI mezzanine, compatible with the ISO_SPI motherboard and route LVDS to the DIOT connector

[image]