sinara-hw/AUX_PSU #26: New revision of AUX_PSU - discussion and fix list

mkiepiela, 2024-01-22:
We are going to make new revision of AUX_PSU. Issues to fix:

- [ ] #10 
- [ ] #11 
- [ ] #12
- [ ] #13 
- [ ] #14 
- [ ] #15 
- [ ] #16 
- [ ] #18 
- [ ] #19 
- [ ] #20 
- [ ] #21 
- [ ] #22 
- [ ] #24 

**Feel free to add any suggestions/issues/ideas for new revision.**

gkasprow, 2024-01-22:
make DIOT variant, expose the EEM connector and we will make Iso-SPI mezzanine.

maciejprzybysz, 2024-03-14:
@mkiepiela https://github.com/sinara-hw/AUX_PSU/issues/27

jbqubit, 2024-06-28:
Looking forward to the updated version. :)

martafafara, 2025-04-24:
V2.0 ready to review

gkasprow, 2025-05-16:
There is one remaining issue from my list:
"add 10uH 1A inductor in series with L12, L11, L10"
Ferrite beads are pure resistances at the switcher frequency, and a lot of SEPIC noise goes through the LDO's poor PSRR at that frequency.

martafafara, 2025-05-22:
@gkasprow I haven't seen it before. inductors added. 
Additionally, power-up delay circuit added. Removed DIOT part break-off milling. Hole in front of  Molex power connector added for easier plug inserting.

jbqubit, 2025-06-05:
Sounds like this is released for manufacturing! Please create a Release tag on github including PDF schematics, layout and BOM. 

Is there a plan for I2C ADC readout support in ARTIQ?